openenergymonitor/emonth2 #6: Adjusting sensitivity of optical sensor

danbates2, 2018-09-09:
Based on my observations and testing done by @kempson
https://community.openenergymonitor.org/t/adjusting-sensitivity-of-optical-pulse-sensor/8399/21

Rather than use the internal pull-up of the atmega, it's probably better to have an external pull-down and 0.1uF capacitor-to-gnd on the interrupt/signal line of the optical sensor. This would help decrease the sensitivity of the sensor and avoid light leakage issues. Also, the cap will help with noisy environments, as it did for my "connecting to RPi project".

Finding an good enough and balanced (reducing sensitivity vs maintaining strong output) pull-down value hasn't been done rigorously yet, when I get another sensor I'll do some checking, I've used a 50k pull-down and that worked fine on the RPi project.

glynhudson, 2018-09-18:
> Rather than use the internal pull-up of the atmega, it's probably better to have an external pull-down and 0.1uF capacitor-to-gnd on the interrupt/signal line of the optical sensor. This would help [...]

Yes, I did test using an external pull-up when in the design phase for the emonTH. However, I found that using the internal pull up provided adequate performance when using our standard optical pulse counting sensor without requiring additional components. Possible if using other sensors or direct pulse counting external components would be desirable. 

Thanks for sharing the results of your testing, it it's proven that an external pull-down will significantly improve performance I would happily consider adding it on a future version of emonTH.

danbates2, 2018-09-19:
I'm using a 45k pull-down on my RPi setup, I'll be able to report on it's accuracy against the meter in about 2 weeks.
Further thoughts below, feel free to read, or wait for my result.

- - -

@Kempson's test result when the sensor is in darkness **is** the proof that a pull-down is better.

> Using a milliameter connected from IRQ to ground, with the sensor in darkness, 0.08 mA of current was drawn. Voc = 0.745 V therefore the internal source resistance = 9.3 k ohm (ignoring meter internal impedance). Similar source impedance is seen if testing from IRQ to Vcc.

The sensor is not able to sink very much current in darkness, and only pulls Voc down to 0.745V. 

> Using a 1 k ohm resistor connected from IRQ to ground, with the sensor illuminated, 2.58 mA of current was drawn. Voc = 3.333 V therefore the internal source resistance = 292 ohm.

His test in bright light shows the sensor is able to output 2.58mA through a 1k resistor, thus if a pull-down was used in the order of 30k the sensor should easily be able to 'drive up' to 3.3V.

I believe the 0.1uF cap works for a few possible reasons:
. light leakage.
. lighting can flicker.
. EM noise, especially from mains cabling.
. Noise at the device from radio, RPi, emonTH etc..