openenergymonitor/emonth2 #15: Custom encryption key

jonathandreyer, 2023-02-20:
Hello, I have implemented an option to build the firmware with a custom encrypted key. It is possible to see the modification [here](https://github.com/jonathandreyer/emonth2/commit/e47dab44f89eb157c9354e1c88b631e73db9302b). Are you interested to integrate it into the repository?
Best regards,
Jonathan Dreyer

TrystanLea, 2023-02-27:
Thanks @jonathandreyer yes happy to merge

jonathandreyer, 2023-02-27:
It is a pleasure @TrystanLea . I have created the linked PR.

Maybe, a cleanup of the code will be useful. When I save the code, some IDE (like VS code or Atom), remove excess space automatically. It will also be nice to have the CI to automatically build the firmware (to each commit) to be sure, there is no compilation error.

PR #16 (2023-03-03): Add possibility to use custom encrypted key
Here is the PR with the new functionality. This PR close #15.
closed 2023-03-03 by TrystanLea